retrospy/RetroSpy #224: List of used pins

DieKatzchen, 2022-01-13:
Making a Teensy based RetroSpy, want to use dsub connectors and the MODE_DETECT option to make switching consoles easier, but I'm not sure how many pins are needed/which pins are used. Is there a list anywhere or do I just have to go through the entire config options and check them off one by one?

zoggins, 2022-01-13:
everything, but Neo Geo and V.Smile  are contained within these 8 pins.  The odds are those are all you ened.

#define DIGITAL_PIN_02	  
#define DIGITAL_PIN_14	  
#define DIGITAL_PIN_07	   
#define DIGITAL_PIN_08	   
#define DIGITAL_PIN_06	   
#define DIGITAL_PIN_20	   
#define DIGITAL_PIN_21	   
#define DIGITAL_PIN_05	  

Neo Geo also uses these pins
#define DIGITAL_PIN_16	   
#define DIGITAL_PIN_27	   
#define DIGITAL_PIN_19	   
#define DIGITAL_PIN_18	   

and V.Smile uses 
#define DIGITAL_PIN_9

DieKatzchen, 2022-01-13:
So basically those 8, GND, and then I can also throw in all of the analogs (for future proofing). A DB25 should be sufficient and I have plenty on hand.

zoggins, 2022-01-13:
Yes, those 8 + GND and the analogs for switching and you should be good for almost anything

zoggins, 2022-01-13:
I forgot to add Wii uses 18 & 19 so you might want to throw those on

zoggins, 2022-01-19:
Closing, feel free to reopen if you have more questions

closed 2022-01-19 by zoggins