retrospy/RetroSpy #226: Generic PC Controller triggers D-Pad on Max X/Y Range

goyney, 2022-01-25:
When utilizing the generic PC controller, if I set up my configuration to watch up/down/left/right and the x/y axis, the D-Pad indicators are triggered with the x/y axis are push to max range.

```xml
<button name="up" image="D-down-up.png" height="17" width="14" y="60" x="30"/>
    <button name="down" image="D-down-up.png" height="17" width="14" y="90" x="30"/>
    <button name="left" image="D-left-right.png" height="14" width="17" y="76" x="14"/>
    <button name="right" image="D-left-right.png" height="14" width="17" y="76" x="43"/>

    <stick xname="x" yname="y" image="joystick2.png" x="94" y="31" width="50" height="50" xrange="30" yrange="30" yreverse="true" />
```

https://user-images.githubusercontent.com/1015040/151041024-6a5f7ce5-5c5b-4542-9c7b-4e9894615d50.mp4

If there is a configuration option to prevent this I have missed, I apologize and would love to see documentation updated to capture this case.

zoggins, 2022-01-25:
What type of controller are you using?

goyney, 2022-01-25:
A generic USB N64 controller. When opening the Game Controller settings in Windows, the POV (D-Pad) keys operate separately from the analog stick in the preview window there (meaning the issue only presents itself within RetroSpy).

zoggins, 2022-01-25:
Do the up/down/left/right keys work correctly by themselves in RetroSpy?

goyney, 2022-01-25:
Yep, I can press any of the D-Pad buttons and they show fine. The analog stick also registers fine. It's only when the analog stick is push to the max/min of a specific axis that it lights up the D-Pad button as well for some reason.

zoggins, 2022-01-25:
And you are using “PC Generic Gamepad” from the drop down?  I think I might see the problem.

goyney, 2022-01-25:
Yes, that is correct.

zoggins, 2022-01-31:
What controller are you using?

goyney, 2022-01-31:
It is a Retro Power controller from this pack: https://www.amazon.com/gp/aw/d/B07491YH2R/ref=cm_cr_arp_mb_bdcrb_top?ie=UTF8

zoggins, 2022-01-31:
What do the buttons and sticks do in the Skin Helper skin under PC generic Gamepad

goyney, 2022-01-31:
The Dpad only lights up the pad buttons. The analog stick moves the analog bars, however when the analog stick is maxed out in a direction, it also lights up the same Dpad direction.

zoggins, 2022-01-31:
https://github.com/retrospy/RetroSpy/releases/tag/v5.0.4  should solve your problem.  Let me know if it works.

goyney, 2022-01-31:
That did it! Thanks so much for the help and quick turnaround. Your work is much appreciated! 😄

closed 2022-01-31 by zoggins