retrospy/RetroSpy #228: Dualshock 4 Incompatiability with RetroSpy and CronusMax Plus

ThePackle, 2022-03-05:
Hello!

Just as an intro, the CronusMax Plus is an adapter for consoles that allows you to plug a different controller into a console then what was intended - in my case, I use it to use a Dualshock 4 on an Xbox 360. While it does have capabilities to perform macros and turbos and whatnot, using different controllers is a huge benefit to me. I've used the DS4 with my CronusMax Plus for about five-ish years now with no huge problems.

My issue, now, is that neither of my Dualshock 4s will not work with this setup - the CronusMax just doesn't recognize it as a controller (although the inputs do show up on NintendoSpy). However, I have tried this with a Dualshock 3 controller, Nintendo Switch Pro controller, and a random gamepad, and all three were recognized just fine and performed as intended. For whatever reason, the DS4 just doesn't get passed through properly enough for it to be recognized - though that is all 100% speculation on my end, since I know this is a very strange and very unique use-case.

I am utilizing the latest version of RetroSpy and the Beaglebone's firmware for this.

If this isn't figured out, it isn't the end of the world - I can always fall-back onto the DS3 with little problems.

Thank you for your time!

zoggins, 2022-03-05:
I think I know why this is happening.  Which end do you have the RetroSpy hardware hooked up to?  The Xbox 360 side or the Dual Shock 4 side?  Does it not work on both sides, or have you only tried one side.  I am pretty sure I know why it won't work on the DualShock 4 side, which I will look at.  I am not even sure if it would work on the Xbox 360 side of the CronusMax.

ThePackle, 2022-03-05:
I may be misunderstanding what you mean, so I apologize. I could make a bad paint drawing if you need, but this is a rough layout:

360 <-- Cronus <-- RetroSpy <-- DS4

Keeping the same layout and as I mentioned, a DS3 and Switch controller will work just fine. And, by both sides, do you mean just reversing the inputs onto the Beaglebone?

zoggins, 2022-05-11:
Sorry for the delay.  I have this fixed in a test build.  I will have something out in a day or two.

zoggins, 2022-06-03:
This will be in the next release

closed 2022-11-23 by zoggins