retrospy/RetroSpy #229: Unable to see input when using  LGT8F328P-LQFP32 MiniEVB (Arduino nano clone)

Quotad, 2022-03-17:
Grabbed a few of [these ](https://www.aliexpress.com/item/1005003703561677.html)to try them out with RetroSpy. They don't seem to work with latest FW and RetroSpy release but I can get them working with NintendoSpy just fine. Any assistance on what I can check, change and try to get these working with RestroSpy would be greatly appreciated

zoggins, 2022-03-17:
What console are you trying to use and by doesn't work do you mean there are just no inputs on the display at all?  I will order some and figure out what the problem is.

Quotad, 2022-03-21:
> What console are you trying to use and by doesn't work do you mean there are just no inputs on the display at all? I will order some and figure out what the problem is.

Sorry for delay. Trying to use with N64. The input viewer fails to log any input from the controller. I have also tested on a different clone but getting same results (NintendoSpy still works fine).

Firmware seems to be flashing to microcontrollers just fine, no errors in Arduino IDE so not too sure what the issue could be

zoggins, 2022-03-21:
No worries.  And in the Arduino IDE you have Tools -> Board -> Arduino AVR Boards -> Arduino Nano selected?  There is conditional compile logic for different boards in the N64 firmware.  I am guessing you are ending up in the wrong section, but I am not sure how yet.

Quotad, 2022-03-21:
This is the board I select when using a "standard" nano clone
[image]

This is the board I select using the LGT8F328P-LQFP32 MiniEVB board
[image]

zoggins, 2022-03-21:
Ok that explains why the LGT8F328P-LQFP32 MiniEVB isn't working, since the code sees the that board as unknown it will default to the firmware that doesnt' do anything other than compile.  Why a standard Nano isn't working is a bit of a mystery still.

zoggins, 2022-03-21:
What standard Nano's are you using?

Quotad, 2022-03-21:
[These](https://www.aliexpress.com/item/1005001694457621.html) are the other nano clones I use

zoggins, 2022-05-03:
[firmware.zip](https://github.com/retrospy/RetroSpy/files/8608136/firmware.zip)

This firmware should fix the problem with the LGT8F328P-LQFP32.  I have only physically tested it with the N64, but it SHOULD work on anything the normal nano works on. I am still investigating why the 168P nano isn't working.

zoggins, 2022-05-03:
I should add that my fix only works if you run the LGT8F328P-LQFP32 at 16Mhz, which unless you modified the Arduino config should be the default.

zoggins, 2022-05-03:
Unfortunately I can't reproduce the problem with the standard Nano.  I bought several from the link you sent and they all seem to work fine with the current release.

zoggins, 2022-05-03:
Are you using the current release 5.0.4?

zoggins, 2022-06-03:
Closing.  Let me know if you are still having issues

closed 2022-06-03 by zoggins