retrospy/RetroSpy #231: Support Sending Data as Bytes, instead of ASCII Bits: Nicohood Nintendo Library: Rebase from nintendospy?

Skuzee, 2022-05-15:
Hello!
Currently using [https://github.com/NicoHood/Nintendo](Nicohood's nintendo lib) to communicate with n64 and gamecube controllers.
The data is stored in bytes, and are a different order than the nintendospy format. The master branch of nintendospy has support for sending the data as whole bytes instead of the typical ascii bits. See [https://github.com/jaburns/NintendoSpy/blob/master/Readers/GameCube.cs](Nintendospy Gamecube Reader) 

I am pushing the limit if what my arduino atmega32u4 can do and sending the input data as ascii is 8x less effective and is causing issues for me, so I can't use retrospy at the moment. nintendospy has not released a compiled binary since 2014 so it's a hassle for some users to compile their own binaries.

Is it possible to rebase retrospy off the current nintendospy master branch that has nicohood support? or add the nicohood support to retrospy? I believe a lot of the code is in NintendoSpy/blob/master/Readers/GameCube.cs (linked above) but I'm not sure what other changes to other file are necessary otherwise I would help merge that feature.

Thanks for your help!
~Angst

Skuzee, 2022-05-15:
Here's the specific commit. It appears the only changes are in the gamecube reader. 
https://github.com/jaburns/NintendoSpy/commit/0e557d9fae47f9d684c82309ada8c8822c591d0e

In `ControllerState ReadFromPacket` is a case select that supports a packet size of 8 (instead of the usual 64)

Skuzee, 2022-05-15:
On closer inspection I think I might be able to add support my self, so I've fork retrospy to give it a try. My question about rebase-ing still stands; is it possible to rebase retrospy off the existing nintendospy? If not, I'll just make the changes that I need manually and request a merge. 💛

Skuzee, 2022-05-15:
I was able to add support for nicohoods 8byte format using the code from the commit from the master branch. I also added DTR and RTS handshakes to SerialMonitor.cs to improve the connection reliability with my atmega32u4. We found that opening/closing the arduino IDE serial monitor would close the serial connection and NintedoSpy was unable to reconnect without initiating it's own handshake.

See pull request https://github.com/retrospy/RetroSpy/pull/232

zoggins, 2022-06-03:
This has been checked into master

closed 2022-06-03 by zoggins