retrospy/RetroSpy #233: OG Xbox - Sync issue

stylonide, 2022-05-19:
Hi there,

I recently set up the Beaglebone Black (model BBONE-BLACK-4G) according to the instructions provided here (https://retro-spy.com/wiki/beaglebone-black-getting-started/) in order to show inputs from an OG Xbox S controller. After completing the setup, I began testing and found that while the d-pad and buttons responded correctly, both analogue sticks suffered a delay/sync issue wherein a character would not respond quickly enough to a sudden change in direction. 

I've attached a video at the bottom of this post demonstrating the issue with me spinning the sticks, though this also happens when quickly moving the stick to other positions (such as moving from front to back). I've also tested this with other Xbox controllers as well as in other Xbox games and found the issue persistent.

I've performed some testing to identify a possible bottleneck in my USB setup chain: I removed the beaglebone from the chain and went Xbox controller > USB male > USB female > Xbox console and found no such delay issues. I've run speedtests and found no deviation in speeds, as well as pinging the beaglebone and receiving a result of 1 ms. 

I've also tested the setup with a nintendo switch using a pro controller via wired connection, performing the same test and discovered no issues. I've also attached a video of that to the bottom of the post.

I'm unsure as to what could be causing this issue, any guidance or feedback is appreciated. If I can provide further details please don't hesitate to ask :)

Cheers

https://user-images.githubusercontent.com/73919279/169212767-7465f856-8080-4a7d-a886-4b5688e5c7c3.mp4

https://user-images.githubusercontent.com/73919279/169213483-928f2eda-08e7-4cdf-abfd-cb8a92452fad.mp4

zoggins, 2022-05-19:
Thanks for the heads up. I will take a look.  Do you notice it happens immediately on start or does it slowdown over time?

zoggins, 2022-05-19:
I ask because the switch had a similar problem, but in that case it slowed down over time.  Also when you move the analog sticks do the digital buttons lag with the lag of the sticks?

zoggins, 2022-05-19:
Also I am sure you tried multiple games, but what is one I can try to reproduce the problem?

stylonide, 2022-05-19:
I appreciate the quick response! Sorry for taking a moment on my end, I wanted to provide as thorough an answer to your questions as I could. I just ran a check by restarting both the Xbox and the Beaglebone, the delay issue is still present. All buttons themselves are responsive, though it appears that the inputs are being queued (I've included some footage of Crash Twinsanity and Sonic Heroes where I attempt to jump while moving Crash/Sonic in a circle. What should happen is they jump while turning, instead the inputs are queued until their running stops. The same happened with Fry's gun in the Futurama clip above). Of the games I tested I used Futurama, Crash Twinsanity and Sonic Heroes. I just gave it a go with Halo (since it's a fairly common title) and it persists there too. 

(Apologies for the quality of both files and the abrupt start in the Heroes clip, I did a very quick and dirty job in Handbrake to get them under the 10 MB limit) 

https://user-images.githubusercontent.com/73919279/169219543-ad0e52da-85de-48f4-95ec-706e12a4fb7a.mp4

https://user-images.githubusercontent.com/73919279/169219658-12ea0eac-8de6-43bf-b064-a2a3b327947a.mp4

zoggins, 2022-05-19:
No worries.  The issue sound like something I had to fix with switch, but I have never encountered it on the Xbox.  I will investigate.

stylonide, 2022-05-19:
Fantastic, thanks for your insight and help. I'll have an eye on the thread here so let me know if there's anything else I can provide :)

stylonide, 2022-05-21:
Some further discoveries-- I've flashed my firmware back to version 3.2 (using the "BBB-blank-debian-9.9-retrospy-iot-armhf-2019-08-01-4gb.img.xz" file from the 7th August 2019 release) and am not currently running into this issue. I'll continue testing with this firmware and let you know if anything comes up.

stylonide, 2022-05-21:
Misclicked a button there, please disregard the above closing and reopening!

zoggins, 2022-05-21:
Did you try any in between the latest and 3.2?

zoggins, 2022-05-21:
I think I might know what is wrong.  I will get the Xbox hooked up this weekend to verify

stylonide, 2022-05-22:
I've only tried 3.2, I used it for the test since it was the earliest mention of Xbox support that I could find. Thanks for taking a look, if there are any other releases you'd like me to test let me know

zoggins, 2022-05-22:
If you get a chance try this one from the 4.4 release.  If its what I think it is, this one should work.

BBB-blank-debian-9.9-retrospy-iot-armhf-2021-07-11-4gb.img.xz<https://github.com/retrospy/RetroSpy/releases/download/v4.4/BBB-blank-debian-9.9-retrospy-iot-armhf-2021-07-11-4gb.img.xz>

stylonide, 2022-05-22:
Just got a chance to test the 4.4 release, unfortunately the input delay issue is still present in an identical fashion to the latest. Are there any others I should test?

zoggins, 2022-05-22:
The next one to test would be the one in Release 3.5 (BBB-blank-debian-9.9-retrospy-iot-armhf-2019-11-02-4gb.img.xz<https://github.com/retrospy/RetroSpy/releases/download/v3.5/BBB-blank-debian-9.9-retrospy-iot-armhf-2019-11-02-4gb.img.xz>).  Thanks a ton for helping me track this down.

stylonide, 2022-05-22:
Happy to help! It's a bit late where I am but I'll have a response on release 3.5 in the coming day.

zoggins, 2022-05-22:
Out of curiousity when you go to an old firmware are you still using the newest UI?

stylonide, 2022-05-23:
Just got in and have some time to test 3.5-- I have been using the latest UI on all builds. Happy to test different releases with the matching UI if needed

zoggins, 2022-05-23:
I spent the day hunting down the commit that is causing the problem.  I found the commit and am working on figuring out why its broken.  I should have a working build in a day or two.  Thanks for all the help and letting me know of the problem!  I really appreciate it.

stylonide, 2022-05-23:
Fantastic! Glad I could be of assistance, keen to check out the new build when it's ready :)

closed 2022-05-23 by stylonide