retrospy/RetroSpy #238: PS4 & PS5 controllers as generic usb gamepad registering left stick with d-pad

developwisely, 2022-06-23:
I've been creating a ton of skins for RetroSpy lately.  Mostly creating generic gamepad skins and MiSTer skins.

I've come across a problem with PS4 and PS5 controllers, where the left analog stick also registers as d-pad when you use it as type generic.

[image]

**type generic settings**
```
<!-- Left Analog -->
<stick xname="x" yname="y" image="analog.png" x="75" y="75" 
          width="32" height="32" xrange="12" yrange="12" yreverse="true" />

<!-- Right Analog -->
<stick xname="z" yname="rz" image="analog.png" x="213" y="75" 
          width="32" height="32" xrange="12" yrange="12" yreverse="true" />
```

**type mister settings**
```
<!-- Left Analog -->
<stick xname="x" yname="y" image="analog.png" x="75" y="75" 
          width="32" height="32" xrange="12" yrange="12" yreverse="true" />

<!-- Right Analog -->
<stick xname="rx" yname="ry" image="analog.png" x="213" y="75" 
          width="32" height="32" xrange="12" yrange="12" yreverse="true" />
```

A few more bits of info:
- this does not occur with MiSTer type
- this does not occur with NintendoSpy as generic type

If you want to try them or see the xmls: https://github.com/developwisely/squabbler-retrospy-nintendospy-skins

zoggins, 2022-06-23:
I will take a look

zoggins, 2022-11-23:
Which skin are you seeing this with the PS4/PS5 controllers?

zoggins, 2022-11-24:
How are you connecting your controllers to your PC?  Is it through Steam or DS4Windows?

zoggins, 2022-11-24:
I need to know more about your setup.  I am using your "Squabbler - Playstation DS3" skin with a ps5 controller and 2 PS 4 controllers using both Steam and DS4Windows and I cannot repro the issue.

zoggins, 2022-12-05:
closing for now.  If its still happening feel free to reopen

closed 2022-12-05 by zoggins