retrospy/RetroSpy #240: Nintendo Switch Pro Controller: Noisy inputs while being connected for PC game.

Robert-Ordis, 2022-07-07:
https://twitter.com/Robert_Ordis/status/1544706782110818304

* ZL(push like striking)
* Lstick_switch
* Home
* Vibrating

zoggins, 2022-07-07:
You have the controller hooked up via the beaglbone to a PC or a switch?  Please share your exact setup and I will take a look. What game is this?

Robert-Ordis, 2022-07-08:
> You have the controller hooked up via the beaglbone to a PC or a switch?
My PC (Asus Rog flow x11, AMD Ryzen 5900HS, NVidia GTX1650)
[NSW Pro controller] - [BeagleBone] = [USB hub port *2] - [PC]

Game is "Little Witch Nobeta", downloaded from steam, but noisy input has been found before starting the game.

zoggins, 2022-07-11:
Does it work if you plug the controller directly to the PC and select one of the PC generic controllers?

Robert-Ordis, 2022-07-13:
No.
Too bad output.

https://user-images.githubusercontent.com/18342677/178770618-22a211e3-a625-4af0-a589-9aab99c7a30a.mp4

zoggins, 2022-11-23:
Which Skin in the RetroSpy UI are you using with this setup?

zoggins, 2022-11-24:
Ok, the actual issue is that the Switch Pro controller input display goes crazy when plugged into a PC.  The Beaglebone really shouldn't be necessary for something plugged into a PC.  I can definitely reproduce the craziness, so I am tackling that and not hte beaglebone problem.

zoggins, 2022-11-24:
Here is the current workaround that doesn't require the Beaglebone:

1. Install [ViGEm Bus Driver](https://github.com/ViGEm/ViGEmBus/releases).
2. Connect your Nintendo Switch Pro Controller with USB cable to PC.
3. Start the driver application [ProconXInputTE_x86.exe](https://github.com/ttsuki/ProconXInputTE/releases/latest) or [ProconXInputTE_x64.exe](https://github.com/ttsuki/ProconXInputTE/releases/latest). _There is no virus if your antivirus complains._
4. Launch RetroSpy.  Select Controller Source "PC 360 Controller" and Viewer Skin "Switch Pro Controller"
5. Hit Go.

The Home and Capture buttons do not currently work, but all the rest of the buttons should work correctly.

closed 2022-11-24 by zoggins