retrospy/RetroSpy #241: 5.0.4 crashing on MiSTer

zoggins, 2022-07-11:
First of all, Kudos to such a cool piece of software and such a neat idea!

I downloaded the newest build of Retro Spy, ran the update script (sucessfully) and tried connecting but as soon as I get the connection after a few second, the controller window flashes for a millisecond and closes down again - is there any reason for this or a fix that needs to be applied? Both my computer and MiSTer FPGA are connected to the same network, my MiSTer is up-to-date, I'm a bit lost on bugfixing or anything related to the topic, as there hardly are ressources!

alexluecke, 2022-07-19:
I am having the same issue. Is there a log I can review to maybe provide some more info?

zoggins, 2022-07-19:
I will get out the MiSTer and see if I can repro.  Last time I couldn't get it to crash.  Can you tell me what controller you are using and what joystick port it is being detected as?

alexluecke, 2022-07-20:
I am using an Xbox One controller. Same thing happens with a PS4 DualShock 4. When I click the "Detect" button controller 0 stays selected.

alexluecke, 2022-07-20:
```
wget -t 3 \
  --no-check-certificate \
  --show-progress \
  https://github.com/retrospy/RetroSpy/releases/latest/download/update-retrospy-installer.sh
--2022-07-19 18:08:59--  https://github.com/retrospy/RetroSpy/releases/latest/download/update-retrospy-installer.sh
Resolving github.com (github.com)... 140.82.114.3
Connecting to github.com (github.com)|140.82.114.3|:443... connected.
HTTP request sent, awaiting response... 302 Found
Location: https://github.com/retrospy/RetroSpy/releases/download/v5.0.4/update-retrospy-installer.sh [following]
--2022-07-19 18:08:59--  https://github.com/retrospy/RetroSpy/releases/download/v5.0.4/update-retrospy-installer.sh
Reusing existing connection to github.com:443.
HTTP request sent, awaiting response... 404 Not Found
2022-07-19 18:08:59 ERROR 404: Not Found.
```

Is this download path supposed to be `https://github.com/retrospy/RetroSpy/releases/download/v5.0.4/update-usb-retrospy-installer.sh` ?

alexluecke, 2022-07-20:
Ah, ok I replaced it with https://raw.githubusercontent.com/retrospy/RetroSpy/master/MiSTer/update-retrospy-installer.sh and re-ran the script and everything worked, so I think it's just a resource thing.

40wattrange, 2022-07-21:
I am getting this same issue - no matter if i run the update script or the update-installer script, same problem.

Takes a moment to connect, get the input display windows for about 1/4 of a second, and then kicks me back to the main GUI.

zoggins, 2022-11-24:
I can't repro this with 5.1, but my guess is its the fast fail timeout feature.  I can increase it, but I would need someone to test it as I can't reproduce the problem.