retrospy/RetroSpy #242: SNES controller on GameBoy Advance wrong inputs displayed

trevynmace, 2022-07-25:
I definitely have kind of an edge-case situation where I'm using a GameBoy Advance Consolizer, which uses a SNES controller for input.
This is what I'm seeing on RetroSpy currently with the Default SNES controller skin:

<input>	| 	<RS display>
Up 		| 	Start
Left 	        | 	Down
Down	|	Up
Right	|	Left
Select	|	Y
Start	        |	Select
Y		|	B
X		|	A
A		| 	Right
B		|	nothing
L		|	X
R		|	L

My setup is just the kit from Etsy for SNES controllers. 

The only thing I can guess is that the clock cycles from the console are mismatched so RS can't know what key was actually pressed but I'm not sure how to address that in code. A custom skin option where I could assign it manually would be great for a case like this too.

zoggins, 2022-07-25:
Interesting. A custom skin would solve part of the problem, but the nothing for the B button is an issue. What consolizer do you have?  You definitely have me intrigued. :)

trevynmace, 2022-07-25:
Nice! :) Yeah the B button was a strange part for sure. This it the consolizer I have: https://www.game-tech.us/product/gbaconsolizer-full-case-2/
The input passthrough works perfectly, not sure if I mentioned that before, just the input display is incorrect.

zoggins, 2022-07-25:
I bought a consolizer, so will get to the bottom of this once it arrives.

zoggins, 2022-11-23:
Well the issue is hte consolizer has a completely different button layout.  And the b button is actually not mapped to anything in the hardware.  A and Y are the "same" buttons.

zoggins, 2022-11-24:
Will be supported in the next release.  It requires a new firmware, but still uses the SNES reader/skins.

trevynmace, 2022-11-24:
Sounds great!! Thank you so much for following up on this :)

closed 2022-12-05 by zoggins