retrospy/RetroSpy #243: SFC Controller Via SNAC On MiSTer Not Being Recognized

MiR4i, 2022-08-19:
I can get RetroSpy to properly recognize both a USB controller (iBuffalo) and my SFC controller through a Daemonbite adapter, but when my SFC controller is plugged in via the SNAC adapter, it is not detected/recognized through RetroSpy.  I have tried both 5.0.4* and 4.4.0 of RetroSpy, but have been unsuccessful in getting the program to detect the controller or its inputs.

I'll be honest, I'm not sure if the SNAC adapter is properly supported by RetroSpy due to the way it interfaces directly with the cores, but having looked at a previous ticket (#194), it seems like it is supported and should be working.  Is there any special setup required for RetroSpy to function properly when a SNAC adapter is in use, or is there a legitimate issue occurring here?

_*5.0.4 is listed on the GitHub release page, but it installs as 5.0.2 and displays 5.0.0.0 in the Help > About menu._

zoggins, 2022-08-19:
Yeah I am bad at versioning :)  It should work with the SNAC adapter.  Is it a stock SFC controller?

zoggins, 2022-08-19:
Can you walk me through your setup (i.e. how everything is plugged in and what hardware you are using)?

MiR4i, 2022-08-19:
> Yeah I am bad at versioning :) It should work with the SNAC adapter. Is it a stock SFC controller?

Yes, it's an official Nintendo SFC controller (Part# SHVC-005).

> Can you walk me through your setup (i.e. how everything is plugged in and what hardware you are using)?

Alright, I'm a fairly new MiSTer user, so apologies if this explanation is a little convoluted, but I'll begin by saying that it's a pre-built MiSTer from Ultimate MiSTer, and it's the full package (digital I/O, USB hub, etc.).

I have a 3-in-1 Daemonbite USB adapter that is detected by RetroSpy as three controllers (0, 1, and 2), which I assume are NES/SNES/Genesis.  If I then plug in my iBuffalo USB controller, a fourth controller is detected after pressing the "Detect" button in the program.  Furthermore, both the SFC controller via the Daemonbite adapter and the iBuffalo USB controller work properly with RetroSpy's MiSTer overlays (I even created one myself).

From there, if I plug in a controller to the SNAC adapter, no additional controller is detected, and if I unplug every USB controller/adapter, then RetroSpy throws an error stating "Couldn't connected [sic] to MiSTer to get connected controllers."

I will add that the SNAC adapter and the controller are properly functioning, though, as I have used it without issue on both the NES and SNES cores of the MiSTer.  However, I did initially struggle to get RetroSpy working with my MiSTer, which may be the root issue here, and I'll do my best to detail what occurred, along with my workaround.

So, my first attempt at getting RetroSpy to interface properly with the MiSTer was unsuccessful using 5.0.4, and I was having the exact same issue as others have outlined in issue #241, where RetroSpy would try to initiate a connection to the MiSTer, but would immediately close the overlay after it appeared on the computer.  I then installed version 4.4.0 since someone else in #241 had suggested that it worked for them, and while the overlay would not immediately close, I was still unable to get any controllers/input to be recognized.  I will add that I had already run the `update-retrospy.sh` script file numerous times on my MiSTer.

Still attempting to troubleshoot this issue, I decided to also install and run the `update-retrospy-installer.sh` file on my MiSTer, as suggested by another user in #241, and that seemed to allow all USB controllers/adapters to be recognized, so I thought all was well.

A few days ago, though, I received my SNAC, and have been unable to get it working with RetroSpy, using either version 5.0.4 or 4.4.0.  Again, USB controllers continue to function without issue, regardless of whichever version I have installed on my computer, but I don't know if my workaround caused me to skip an additional installer script somewhere that's required for the SNAC to work.  It sounds like the SNAC should be supported, so I'd love to get it working with my overlay, and I will try to provide any additional details that you may need.

MiR4i, 2022-08-23:
While waiting to hear back, I decided to completely re-install my MiSTer setup since everything came pre-installed and I wasn't sure if anything custom was done to it.

After the initial installation, I followed the instructions outlined on the main site (https://retro-spy.com/wiki/setting-up-retrospy-for-the-mister/), and here is what I see when I run the `update-retrospy.sh` file on my MiSTer after placing it in the `/Scripts/` folder:

[image]

However, I am, again, hit by the same issue discussed in #241, where the controller overlay on the PC appears for a brief moment before immediately closing and kicking me back to the main GUI of RetroSpy (v5.0.4).  In addition, here is what RetroSpy is detecting when I press the "Detect" button:

1) If only a keyboard is plugged into the MiSTer, nothing is detected by RetroSpy (an error pop-up appears).
2) If I then plug in my 3-in-1 Daemonbite adapter (to any USB port on the MiSTer), the following Controllers are detected: 0/1/2
3) If I then plug in my SNAC to the User I/O serial port, no additional Controllers are detected (RetroSpy still shows 0/1/2).
4) If I then un-plug my Daemonbite adapter, with the SNAC still plugged in, nothing is detected by RetroSpy (an error pop-up appears).

_Note: I am pressing the "Detect" button in RetroSpy every time I make a change to what's plugged into the MiSTer._

At this point, I haven't the slightest clue what I might be doing wrong since this MiSTer install is completely fresh, the setup instructions for RetroSpy are simple to follow, and everything else on the MiSTer is functioning as I'd expect (network, USB controller input, SNAC controller input, etc.).

While I understand these are probably two unrelated issues, are you unable to confirm either the disappearing overlay from #241 or the lack of MiSTer SNAC detection on your end?  Before I implement my workaround to stop the overlay from disappearing immediately (mentioned in my previous post), are there any log files, or other files that I can look at on the MiSTer, or PC, that will give any clues as to what's happening?

zoggins, 2022-08-23:
So I think I see the problem.  You cannot spy on controllers plugged into the SNAC adapter from the MiSTer side of things.  The data from the SNAC adapter never makes it out of the core that is running.  To spy on controllers plugged into the SNAC adapter you need to spy on them the same as you would original hardware (i.e. Arduino and spliced cable).  Sorry for the confusion.

MiR4i, 2022-08-24:
> So I think I see the problem. You cannot spy on controllers plugged into the SNAC adapter from the MiSTer side of things. The data from the SNAC adapter never makes it out of the core that is [...]

Understood, thanks.  I'll either have to live without the overlay, switch back to USB input, or look into the solutions you have available in your store.

closed 2022-08-24 by MiR4i