retrospy/RetroSpy #250: linux version wont run on fedora

ZeFoxii, 2023-01-21:
i tried downloading and installing the .sh, the tar file and following the instructions on the website to no avail it will not run in any way shape or form.

zoggins, 2023-01-21:
Unfortunately, with no explanation from Microsoft, .NET 7.0 does not support Fedora (yet?).  The official docs say its coming, but I have no idea when.  Currently supported distros are CentOS 7 (x86_64), Debian 10 & 11 (amd64, arm64), openSUSE 15 (x86_64) and Ubuntu 18.04 through 22.10 (amd64).  As soon as .NET 7.0 supports Fedora I will get it working.

zoggins, 2023-01-21:
Out of curiosity, what version of Fedora?  I am seeing some ideas of how to get .NET 7.0 working in Fedora that I will investigate.

ZeFoxii, 2023-01-21:
nobara from glorious eggroll, i figured out you can launch the application from the terminal but it wont launch via double clicking.

zoggins, 2023-01-21:
hmmm, that is strange.  I will look into it.

ZeFoxii, 2023-01-21:
Unhandled exception. System.Exception: XOpenDisplay failed
   at Avalonia.X11.AvaloniaX11Platform.Initialize(X11PlatformOptions options) in /_/src/Avalonia.X11/X11Platform.cs:line 57
   at Avalonia.AvaloniaX11PlatformExtensions.<>c__0`1.<UseX11>b__0_0() in /_/src/Avalonia.X11/X11Platform.cs:line 280
   at Avalonia.Controls.AppBuilderBase`1.Setup() in /_/src/Avalonia.Controls/AppBuilderBase.cs:line 303
   at Avalonia.Controls.AppBuilderBase`1.SetupWithLifetime(IApplicationLifetime lifetime) in /_/src/Avalonia.Controls/AppBuilderBase.cs:line 179
   at Avalonia.ClassicDesktopStyleApplicationLifetimeExtensions.StartWithClassicDesktopLifetime[T](T builder, String[] args, ShutdownMode shutdownMode) in /_/src/Avalonia.Controls/ApplicationLifetimes/ClassicDesktopStyleApplicationLifetime.cs:line 208
   at RetroSpy.Program.Main(String[] args) in /home/zoggins/repos/RetroSpy/RetroSpyX/Program.cs:line 12
Aborted
im giving up for now as this is my most current terminal message and im getting too tired to look into it further

zoggins, 2023-01-21:
I will look into it.

zoggins, 2023-01-21:
are you using the gnome or kde version of nobara?

zoggins, 2023-01-21:
I got it working in Fedora 37 and Nobara 37 Official.  Had to make a bug fix to get my controller working, so 6.0.1 works on my test machines.

ZeFoxii, 2023-01-22:
i use gnome with the thunar file manager.

zoggins, 2023-01-22:
Did you use the Npbara-Gnome installer or the Nobara-Official installer?

ZeFoxii, 2023-01-22:
i upgraded from fedora 36 to 37 using the instructions from the discord support. I did use the official gnome iso

zoggins, 2023-01-22:
ok, let me try the gnome iso, i only tried the official iso.

ZeFoxii, 2023-01-22:
i then installed thunder after the fact, i didn't like the changes made in the gnome file manager

zoggins, 2023-01-22:
Does it crash right as you start it?  I cannot seem to repro, let me try installing the file manager.

zoggins, 2023-01-22:
did you install thunar from source?

ZeFoxii, 2023-01-22:
no i used dnf (sudo dnf install thunar) after sync of the database and keys

ZeFoxii, 2023-01-22:
i found a work around that works well for me first i ran ( sudo chmod 755 retrospy ) then ( sudo chmod a+x retrospy )
then i made a sh file with this command (cd "/home/USERNAME/Desktop/RetroSpy-Linux/bin/" && ./retrospy
) and ran it from the desktop as a program.

ZeFoxii, 2023-01-22:
and now it works

closed 2023-09-20 by zoggins