retrospy/RetroSpy #251: MiSTer updater scripts incomplete in 6.0 release

paulofmandown, 2023-01-21:
`update-retrospy.sh` in the latest release contains references to a private repository and expects a Personal Access Token to be hard-coded. It's also missing a `)` on line 12.

paulofmandown, 2023-01-21:
This was using the copy of the script from the `RetroSpy-Setup-x64.exe` installer

zoggins, 2023-01-21:
thanks, looking into it.

zoggins, 2023-01-21:
should be fixed now,

paulofmandown, 2023-01-21:
That did the trick, but I had to remove the old version that was at `/media/fat/retrospy/retrospy` first. The call to wget was not attempting to overwrite that existing version.

I get the below error now when I select `Go!`, and the process exits with no error displayed when I try to detect controllers.

```
Could not load file or assembly 'Renci.SshNet, Version=2020.0.2.0, Culture=neutral, PublicKeyToken=1cee9f8bde3db106'. The system cannot find the file specified.

   at RetroSpy.SSHMonitor..ctor(String hostname, String command, String username, String password, String commandSub, Int32 delayInMilliseconds, Boolean useQuickDisconnect)
   at RetroSpy.Readers.SSHControllerReader..ctor(String hostname, String arguments, Func`2 packetParser, String username, String password, String commandSub, Int32 delayInMilliseconds, Boolean useQuickDisconnect) in D:\pub\src\Repos\retrospy\RetroSpy\RetroSpyX\Readers\SSHControllerReader.cs:line 19
   at RetroSpy.InputSource.<>c.<.cctor>b__109_0(String hostname, String username, String password, String commandSub) in D:\pub\src\Repos\retrospy\RetroSpy\RetroSpyX\InputSource.cs:line 11
   at RetroSpy.SetupWindow.GoButton_Click(Object sender, RoutedEventArgs e) in D:\pub\src\Repos\retrospy\RetroSpy\RetroSpyX\SetupWindow.axaml.cs:line 412
```

The older version I have of RetroSpy functions without issue with the new version of the MiSTer-side process. The older version I have isn't one of the normal releases, but was built from the old mister_support branch at commit e3abdd50aa82392f2bd7a10875b08d29b76955f1.

zoggins, 2023-01-21:
Well that is concerning, let me take a look.

paulofmandown, 2023-01-21:
It works if I build it myself from master, interestingly

zoggins, 2023-01-21:
It looks like my build machine is pulling the wrong version of the dll.  Yeah building it on my dev box works, so I gotta figure out where in the heck my build machine is pulling the older version of the assembly from.

zoggins, 2023-01-21:
6.0.2 should fix this

paulofmandown, 2023-01-21:
confirmed working for me

closed 2023-01-21 by paulofmandown