retrospy/RetroSpy #255: Issue with Pop!_OS

FelixHelixTM, 2023-02-02:
The program itself opens and runs, however, on Pop!_OS 22.04 I am unable to get RetroSpy to detect anything from the COM ports, even after giving the proper user permissions as stated in the guide. While Pop!_OS 22.04 is based off Ubuntu 22.04, I'm unsure if there's something causing a compatibility issue or not, and if RetroSpy simply doesn't work correctly with Pop!_OS, or if it's user error!

zoggins, 2023-02-02:
You rebooted after permissions?  I will take a look.

FelixHelixTM, 2023-02-02:
[image] I have rebooted, and this is all the appears for me, also, I am using an eletromodder v2. Not sure if that could be causing any issues, but it does work with the windows version.

zoggins, 2023-02-02:
Try "Options -> Don't Filter COM ports"

zoggins, 2023-02-02:
I'm installing Pop!_OS right now to see what I can see.

FelixHelixTM, 2023-02-02:
Okay! I just tried that but still the same result. :(

zoggins, 2023-02-02:
sudo apt-get remove brltty
reboot

Should fix your problem

FelixHelixTM, 2023-02-02:
Thank you! It does work now!

zoggins, 2023-02-02:
I added Pop!_OS to the optional step that says to remove that package.

closed 2023-02-02 by zoggins