retrospy/RetroSpy #256: Bad Interpreter - Cannot Connect

Chromagram, 2023-02-14:
Hello,

I've tried following the steps on the site about setting this up on the MiSTer. When I run the update-retrospy.sh script, it has a line that says /tmp/update-retrospy-installer.sh: /bin/bash^M: bad interpreter: No such file or directory.

The line after still says Installation complete. However, no matter how many times I try, the viewer software on Windows will not connect, and it just says "Cannot connect to -MiSTer IP-".

Is there something I'm missing? Only thing I can think of is my PC is connected to Ethernet (which appears in Windows as the same network name as the MiSTer's network) and the MiSTer itself is on Wi-Fi (I don't have a second 15 foot long cable to run across my floor for the MiSTer. However, I did try both on Wi-Fi on the same network and it's the same problem.

zoggins, 2023-02-14:
The encoding must have gotten messed up when I uploaded it, I will get that fixed.  Do you have SSH enabled on your MiSTer?  The MiSTer code has nothing to do with the required SSH connection that is made.

zoggins, 2023-02-14:
Also is this the 6.1 release?  Just want to make sure.

Chromagram, 2023-02-14:
Thanks for the prompt reply :)

Yes, this is the 6.1 version, I picked up the newest release from GitHub directly. I believe SSH is enabled by default on MiSTer? I do use Wi-Fi on MiSTer to FTP files and games onto it.

zoggins, 2023-02-14:
I will take a look.

zoggins, 2023-02-14:
Ok the interpreter thing should be fixed now.  I am looking into the connection problem.

Chromagram, 2023-02-14:
Thank you! Appreciate the help a lot.

zoggins, 2023-02-14:
The cannot connect is a bug.  Working on a fix now

zoggins, 2023-02-14:
This should be fixed in 6.1.1

Chromagram, 2023-02-15:
Can confirm it's up and running. Thanks again for the quick reply and fix!

closed 2023-02-15 by zoggins