retrospy/RetroSpy #257: Transparent Viewer Skin?

Chromagram, 2023-02-15:
Heya, I'm wondering if there's a way to have a transparent viewer skin for RetroSpy? It'll need some basic explanation.

Essentially, I stream across a wide number of different consoles, and correct me if I'm wrong, but I can't seem to find basic, non-customized viewer skins for a lot of consoles overall.

What I'd love to see is a skin that has no controller image, it just illuminates pressed buttons and you can move the buttons around yourself to reorganize their placements. This way, RetroSpy still illuminates inputs when pressed, but I can layer the viewer on top of any controller image I want and move the buttons in the viewer myself to line them up.

I remember XPadder on PC showcasing the idea well: https://youtu.be/HEbLO1htCcs?t=31
XPadder is, of course, a keybinding program for your controllers, but the visual idea is there.

Press a button on your controller, it lights up as the input, then you move that over any controller image you want so that when you press it, it lights up the area you want.

I'm asking this because even on MiSTer, I couldn't find any uniformity to any released viewer skins that covered all the consoles I wanted, and commissioning/requesting every single one isn't exactly feasible either. Plus, when I even just tried the built-in SNES controller skin, the buttons I was pushing on my controller actually didn't match on the viewer. I think if this option were made possible somehow, we could all pretty easily adjust our own requirements by overlaying our own controller images.

Please let me know if this is possible. Thank you!

zoggins, 2023-02-15:
Making a single skin that just illuminated buttons for any controller wouldn’t be hard, but right now there is no easy way to move those images around other than editing coordinates in the skin file.  I mean I can start there if that would help.

Chromagram, 2023-02-16:
Even having the option to fiddle with it via coordinates in a file would be leaps, thank you!