retrospy/RetroSpy #263: V.smile controller pin-out.

bzuidgeest, 2023-05-28:
I'm trying to re-use the vtech vsmile (wired) controller with a pi pico. I have a hard time finding the needed information. 

From your sources and some others on the internet I have at least determined that its serial data at 4800 baud. I hoped to get the connector pinout from your page https://retro-spy.com/wiki/vtech-vsmile-on-teensy-getting-started/

But the pictures don't match what is being talked about. The seem to have been combined from multiple articles on similar topics for different systems.

I cannot even confirm the noted pinout for the ps2 style connector that it lists.
I states something about connecting two pins. And the code uses two serial ports as a man in the middle I assume. But I would expect a third ground pin should at least be connected to the controller too, to keep signal from floating. But this isn't noted. The drawing also doesn't match up with whatever measurements I have been able to take. 

It's all very confusing. 

For me it boils down to a single question. Do you happen to know the full pinout of the connector preferably from the perspective of the controller cable?

zoggins, 2023-05-29:
Give me a few days and I will hook up the logic analyzer up and see if I can get a more exhaustive pinout.

bzuidgeest, 2023-05-30:
That would be great!

closed 2023-09-20 by zoggins