retrospy/RetroSpy #264: not working when i use dolphin

DerSeaEagle, 2023-06-14:
i used the dolphin mayflash adapter option but when i use dolphin it dosn't recocnize the input from the gamecube controller and when i try to star retro spy after it theres an error 
also i don't have any special adapter

zoggins, 2023-06-15:
Ok, what controller are you using and how is it connected to your machine.  Also what type of machine Windows/Mac/Linux?

DerSeaEagle, 2023-07-04:
regular gamecube controller and just some random adapter

DerSeaEagle, 2023-07-04:
https://www.amazon.de/Keten-Gamecube-Controller-Adapter-Nintendo-Schwarz/dp/B07L8B9LT8/ref=sr_1_3?__mk_de_DE=ÅMÅŽÕÑ&crid=1TT416W8QKDDW&keywords=gamecube+adapter&qid=1688472804&sprefix=gamecibe+adapter%2Caps%2C168&sr=8-3 
i think it was this one

closed 2023-09-20 by zoggins