retrospy/RetroSpy #268: Transparency Mode w/ Game Capture option in OBS

40wattrange, 2023-09-19:
I saw that the new Input Visualizer by KungFusedMike explained how you can set the transparency in the app itself, then instead of using Window Capture, you can use Game Capture and it's option to capture transparency. 

Is that something that could be brought over to RS so we can just use a transparent PNG file instead of needing to use a defined transparency color? This would make skins look a lot better on edges and free designers from having to shift colors to no impact the skins themselves.

[image]

zoggins, 2023-09-19:
The current issue is Game Capture doesn't show the RetroSpy Viewer window at all, so I have to figure that part out first

zoggins, 2023-09-19:
Actually game capture is just broken on my machine, it will not show any windows.