retrospy/RetroSpy #269: N64 shuts off when arduino is plugged in

SioNisti, 2023-09-28:
tried this on an arduino uno and some fake nano and neither work.

no controller plugged in = works
controller plugged in = works
controller plugged in through extension cable = works
arduino connected to the pc = doesnt work

the weird thing is that it used to work with the same setup.  the arduino is defintely the culprit but no clue what the problem is.
i have also made a SNES input display which has no problems.

see video.  problem at 2:05
https://www.youtube.com/watch?v=jEsKjfFP0YE

zoggins, 2023-10-02:
It sounds like you are shorting the power line to GND when plugged into the Arduino. Are you sure you have GND on the left looking at the end of the cable attached to the controller (with the semi circle up)?   Not sure if that is the problem but it definitely sounds like shorting power to ground, which I have done more than once. :)

zoggins, 2023-10-02:
It might help to send me a pic of your cable end with labels for what you have them wired to

SioNisti, 2023-10-06:
> It might help to send me a pic of your cable end with labels for what you have them wired to

yeah i should have done that immediatly, i checked the cables multiple times with a multimeter.  ill get a pic for you tomorrow

SioNisti, 2023-10-13:
sorry for the long wait i got really lazy. 
but here is a picture of the arduino.
[image]

and here is a beautiful paint drawing with what color cable corresponds to which "hole" in the cable end.  i tested these with my multimeter so they should be right.
[image]

the setup in the picture doesnt power off the console but it doesnt get any input from the controller.
[image]

SioNisti, 2023-10-13:
okay really weird.  i switched the orange and the black cable (black to ground and orange not connected) and it works BUT not on retrospy. only on nintendo spy.  i also have the same problem with my n64 input display that on retro spy it thinks im holding down all the buttons but nintendo spy works fine.

okay reinstalling retrospy fixed it.  that really weird it did work on this very install with my ps2 input display.

zoggins, 2023-10-13:
Trying to think why its working in Nintendospy and not retrospy.  What firmware do you have on the Arduino?

SioNisti, 2023-10-13:
i uploaded the frimware.ino from the broken retrospys firmware folder

closed 2023-11-02 by zoggins