retrospy/RetroSpy #270: A/B and X/Y are inverted using MiSTer

PonchoTaco, 2023-10-05:
Hello,

Having an issue setting this up. I am using an 8bitDo Pro 2 controller that is USB connected to my MiSTer and I'm using X input. RetroSpy can see my controller and everything works except when I push A - RetroSpy highlights B (vice-versa). Same results for the X/Y buttons.

I've gone through my MiSTer button mappings and double checked that all my buttons are mapped appropriately at both the system level and the core level.

zoggins, 2023-10-05:
What skin are you trying to use?  The MiSTer skins are complicated because everyone has so many different controllers.  The easiest thing to do would be to modify the skin to reflect the correct button mappings.  I can do it for you if I know what skin you are using

PonchoTaco, 2023-10-05:
I was just using the default SNES Controller skin to test out the app.

PonchoTaco, 2023-10-05:
I got it figured out. Found the skin.xml and was able to edit it to reflect my buttons

closed 2023-11-02 by zoggins