retrospy/RetroSpy #271: Usb adapter not working with RetroSpy

mbarela82, 2023-10-23:
I'm using a usb adapter with my Mister and RetroSpy doesn't seem to recognize it. I'm trying to use an NES oem controller and it doesn't show up in the skins. It does work with the SNES skin that is available by default but the A and B buttons are reversed.

The adapter I'm using is the 4dapter by Timville on Tindie.

zoggins, 2023-10-23:
What source are you using?  The MiSTer one?  Try this one.  Drop it in the skins directory in the RetroSpy folder in program files.  Say yes when it wants to over write.  Let me know if it works.

zoggins, 2023-10-23:
Found a mistake.  Here is an updated one.

zoggins, 2023-10-23:
Found another mistake.  Try this one.

mbarela82, 2023-10-23:
Thank you for your response. I don't see the file on your reply.

zoggins, 2023-10-23:
[wii-nes_3.zip](https://github.com/retrospy/RetroSpy/files/13078505/wii-nes_3.zip)

mbarela82, 2023-10-23:
Thank you! I'll try it out and let you know.

mbarela82, 2023-10-23:
That works! Thank you so much!

closed 2023-10-23 by mbarela82