retrospy/RetroSpy #273: Random Inputs displayed on controller when in door/screen transitions, or being killed in Super Metroid or Super Mario World.

SOBON2O, 2023-12-16:
I purposely had a SNES/SFC input viewer built for the sole reason of fixing this damn problem. I had been using the programs "Input Viewer", & later "ButtonMash", through USB2SNES (QUSB2SNES now). But it seems there wasn't any reason to get a physical input viewert since I have the same problem!!
It's super frustrating…

zoggins, 2023-12-16:
Are you on a Super NT?  If so, this will likely solve your problem: https://github.com/retrospy/RetroSpy/issues/235.  If not, what hardware/mods are you running.  The problem is a known issue on some people's Super NTs, but I have never been able to reproduce the problem on my Super NT and we have never figured out what causes it, but turning off hotkeys does seem to fix it for most people.

SOBON2O, 2024-01-22:
That is extremely annoying that DOES NOT HAPPEN on a SuperNT. You would think that it would function correctly on legitimate hardware. Because I am not on a SuperNT.

I am playing on a Super Nintendo Jr., and this effect I described happens when using both my SD2SNES (Rev. J), and on a authentic OEM cartridge of either/or Super Mario World or Super Metroid.

I have seen Streamers using USB2SNES, and the add-on application "Input Viewer", or "ButtonMasher" on QUSB2SNES…and they have this same problem I have described…

*In fact, the whole reason I f*cking switched from using QUSB2SNES and Button Masher; to having a physical input viewer device built by someone, WAS TO FIX THIS PROBLEM! Lmao

zoggins, 2024-01-22:
This is the first I am hearing about it happening on original hardware.  Standard OEM controller?  I genuinely want to help, but I have to figure out what's different about your setup and mine (and many other people who don't encounter this problem).  SNES  so the smaller top loader.  That is what I have.  Any mods to the console like RGB or HDMI?  Who built your input display?  Do you know if the GND wire is connected? I'd even be willing to send you one of mine to see if it has the same issue.  Is your streaming PC a laptop or desktop?

SOBON2O, 2024-02-14:
I use standard OEM Super famicom controller.
I have a RGB mod & s-video mod. The console outputs component video at the moment to a Sony 14" PVM.
I stream by outputting the video to a genuine OSSC, and then I use Startech USB3HDCAP (with the Thrillness custom drivers) as the capture card for the final output. 
I have tried both Super Nintendo, and Super Famicom controllers, using the USB3HDCAP only without the OSSC, Etc.
The problem still persists.

zoggins, 2024-02-14:
Who made your RetroSpy hardware?  Do you know what version firmware is on it?  What microcontroller it is using?