retrospy/RetroSpy #278: DIY PS1 cable: PSX Pin 2 to Arduino Uno Pin 5 makes controller unresponsive

tyisafk, 2024-03-14:
Same as title.  I triple checked the continuity on both ends in my DIY cable.  Everything matches up but when I connect the corresponding wire to digital pin 5 on my Uno, the controller becomes unresponsive.

Controller: PS1 Digital and PS2 Dual Shock
Console: PS2 90k Slim
Arduino: Elegoo Uno R3 (I've used this Arduino for N64, Gamecube and SNES DIY cables, they all work.)

zoggins, 2024-03-14:
Usually when that happens it means something is shorting to GND.  Not that colors should be trusted, but what colors of wire go to which PS pins?

tyisafk, 2024-03-14:
I was just about to edit my comment.  My arduino's digital pin 5 is shorted to ground for whatever reason.  Checked the cable and it's fine.

zoggins, 2024-03-14:
No worries.

tyisafk, 2024-03-14:
Closing.  Replacement Arduino solved the issue.