retrospy/RetroSpy #284: [BUG] Raphnet N64/GC to USB Adapter

Dignity1337, 2024-06-29:
Hi. I'm using a Raphnet N64/GC to USB adapter. If I then choose the Controller Source "Generic PC Gamepad" I get an error. The skin does not matter. I get this error for all skins. And also on both ports (Controller 0 or Controller 1). It does not matter if I connect a N64 oder GC controller. It always says this (and sometimes it crashes); 

Invalid name [Z-Achse]. Must be in [X;Y;Z;RotationX;RotationY;RotationZ;Sliders0;Sliders1;PointOfViewControllers0;PointOfViewControllers1;PointOfViewControllers2;PointOfViewControllers3;Buttons0;Buttons1;Buttons2;Buttons3;Buttons4;Buttons5;Buttons6;Buttons7;Buttons8;Buttons9;Buttons10;Buttons11;Buttons12;Buttons13;Buttons14;Buttons15;Buttons16;Buttons17;Buttons18;Buttons19;Buttons20;Buttons21;Buttons22;Buttons23;Buttons24;Buttons25;Buttons26;Buttons27;Buttons28;Buttons29;Buttons30;Buttons31;Buttons32;Buttons33;Buttons34;Buttons35;Buttons36;Buttons37;Buttons38;Buttons39;Buttons40;Buttons41;Buttons42;Buttons43;Buttons44;Buttons45;Buttons46;Buttons47;Buttons48;Buttons49;Buttons50;Buttons51;Buttons52;Buttons53;Buttons54;Buttons55;Buttons56;Buttons57;Buttons58;Buttons59;Buttons60;Buttons61;Buttons62;Buttons63;Buttons64;Buttons65;Buttons66;Buttons67;Buttons68;Buttons69;Buttons70;Buttons71;Buttons72;Buttons73;Buttons74;Buttons75;Buttons76;Buttons77;Buttons78;Buttons79;Buttons80;Buttons81;Buttons82;Buttons83;Buttons84;Buttons85;Buttons86;Buttons87;Buttons88;Buttons89;Buttons90;Buttons91;Buttons92;Buttons93;Buttons94;Buttons95;Buttons96;Buttons97;Buttons98;Buttons99;Buttons100;Buttons101;Buttons102;Buttons103;Buttons104;Buttons105;Buttons106;Buttons107;Buttons108;Buttons109;Buttons110;Buttons111;Buttons112;Buttons113;Buttons114;Buttons115;Buttons116;Buttons117;Buttons118;Buttons119;Buttons120;Buttons121;Buttons122;Buttons123;Buttons124;Buttons125;Buttons126;Buttons127;VelocityX;VelocityY;VelocityZ;AngularVelocityX;AngularVelocityY;AngularVelocityZ;VelocitySliders0;VelocitySliders1;AccelerationX;AccelerationY;AccelerationZ;AngularAccelerationX;AngularAccelerationY;AngularAccelerationZ;AccelerationSliders0;AccelerationSliders1;ForceX;ForceY;ForceZ;TorqueX;TorqueY;TorqueZ;ForceSliders0;ForceSliders1]

   at Vortice.DirectInput.IDirectInputDevice8.GetFromName(String name)
   at Vortice.DirectInput.IDirectInputDevice8.GetObjectPropertiesByName(String name)
   at RetroSpy.Readers.GamepadReader..ctor(Int32 id) in D:\src\Repos\retrospy\RetroSpy\RetroSpyX\Readers\GamepadReader.cs:line 64
   at RetroSpy.InputSource.<>c.<.cctor>b__132_36(String controllerId, Boolean useLagFix) in D:\src\Repos\retrospy\RetroSpy\RetroSpyX\InputSource.cs:line 71
   at RetroSpy.SetupWindow.GoButton_Click(Object sender, RoutedEventArgs e) in D:\src\Repos\retrospy\RetroSpy\RetroSpyX\SetupWindow.axaml.cs:line 434

Anyways I tried the same thing in NintendoSpy.  I select "Generic PC Gamepad" and it works fine (ok need to do a custom skin for this but it works). So this problem seems to be RetroSpy exclusive.

Hope that someone could  fix this.
Have a nice day.
- Jannis

zoggins, 2024-06-29:
I will take a look, I am pretty sure I know what is wrong.

Dignity1337, 2024-06-29:
Thanks for the fast reply zoggins. Would be really awesome if this could be fixed with the next release. RetroSpy is just so much cooler than Nintendospy (which I need to use until then). By the way I noticed that it says "Invalid name [Z-Achse].". I'm from germany on a computer that is set to german language. "Z-Achse" is german for "Z-Axis". So maybe it's a language setting related problem. I don't know if something like this could actualle bbe the case.

zoggins, 2024-06-29:
I am building a test build right now that might solve your problem.  I don't have a German computer so I can't test it, will update this thread when the build is uploaded

Dignity1337, 2024-06-29:
Awesome. Thank you. If you upload the test build I'll let you know if it works. Will go to bed now but be sure that I'll definitely test it within 24 hours.

zoggins, 2024-06-29:
Try this: https://github.com/retrospy/RetroSpy/releases/download/v6.4.9/RetroSpy-Setup-x64.exe

Dignity1337, 2024-06-30:
> Try this: https://github.com/retrospy/RetroSpy/releases/download/v6.4.9/RetroSpy-Setup-x64.exe

Sadly it's not working. During the installation it fails to install the Arduino Srl driver and the Linino (usbser) driver. Not sure if this is part of the problem. After installing and starting RetroSpy it sometimes opens the Retrospy application but sometimes I have to try multiple times before it starts, Also the error changed. It's basically the same error but with different lines:

> Invalid name [Z-Achse]. Must be in [...]

Dignity1337, 2024-06-30:
Ok now it's getting kind of weird. Since your build most of the times doesn't start properly I played around a bit with the current release build. Like I mentioned Z-Achse is a german word so I downloaded the Englisch (United States) language packet for Windows 10 and then set my system to english. And just like that everything works fine. If I switch back to german the error of course shows up again. So the problem is definitely related to the language. What I don't get is that NintendoSpy works fine. This problem only occurs with RetroSpy.

zoggins, 2024-06-30:
Let me try installing the German language pack.  Not sure why it's not starting up consistently.  The driver failing is normal, not an issue.  The reason NintendoSpy is working is because the controller handling code uses an older version of the library that apparently handles this where the newer version does not.

zoggins, 2024-06-30:
Ok, I installed the German language pack and was able to reproduce the problem.  Try this build: https://github.com/retrospy/RetroSpy/releases/download/v6.4.9/RetroSpy-Setup-x64.exe

Dignity1337, 2024-06-30:
Well now it doesn't start at all anymore. I tried rebooting my computer and I also tried running it as administrator. But RetroSpy won't start. Could there be anything wrong with the installer?
The current release starts without problems. But I didn't use any installer for it. Just downloaded it and then I can start RetroSpy by double clicking the application. So might there be something wrong with the installer? I really don't know how to get it working.

zoggins, 2024-06-30:
https://github.com/retrospy/RetroSpy/releases/download/v6.4.9/RetroSpy-Windows-x64.zip

zoggins, 2024-06-30:
You were running 6.4.8 before?

zoggins, 2024-06-30:
What version of windows?

Dignity1337, 2024-06-30:
I think it's version 6.4.8. Dows RetroSpy displays the version somewhere? Just so that I can verify. It's the version from yesterday. I downloaded it yesterday and when it didn't work I immediately opened this issue.
Thanks for providing the zip file but this one also doesn't start at all. So the installer seems to be correct.
I'm running Windows 10 Pro 64 bit.

zoggins, 2024-06-30:
Well the problem seems to be my build server.  I can launch on my dev box, but not the version that my build server creates.  I am working on it

zoggins, 2024-06-30:
Try this: https://github.com/retrospy/RetroSpy/releases/download/v.6.4.9.1/RetroSpy-Windows-x64.zip

Dignity1337, 2024-06-30:
Wow this one now works completely fine. RetroSpy starts and the error is gone. I just copied my altered N64 skin that I had in NintendoSpy (basically the N64 skin only with buttons remapped). Everything is working and all Buttons are correctly recognized.
Thanks a lot.

Dignity1337, 2024-06-30:
It's really awesome that you still keep this project going and offer such a great support. Thanks.

closed 2024-08-20 by zoggins